vtattyba/NASA-Micro-G-NeXT #21: Proper install of KerberosSDR software

teresabompczyk, 2020-06-01:
Get officially-released Kerberos software uploaded to SD card for Pi, and verify proper installation.

teresabompczyk, 2020-06-02:
.img file burned to SD, connection to Kerberos established and software running. Receiving spectral graphs unable to display and hence calibration is set back. Recent forum posts discuss issues with Pi 4, this needs to be looked into and resolved ASAP. Possible resolutions include transitioning down to 3rd gen. Pi or purchasing another company's SDR (an alternative has been located).

teresabompczyk, 2020-06-03:
Posted to rtl-sdr blog, following with notifications if another user responds.

aagarden, 2020-06-08:
I think this has a tutorial on software installation procedure, I also put it in my doc.
https://www.rtl-sdr.com/tag/kerberossdr/

teresabompczyk, 2020-06-08:
rtl-sdr blog responded, thinking it may be a power issue? Using wall adapter rated for 3A, 5V and generic Micro USB cables (which might be the issue if they are not transferring data at the speed needed). Following up with blog post to get some 'high-quality cable' suggestions, and will test further.

closed 2020-11-17 by tamaghan